FreeRTOS/iot-reference-stm32u5 #61: build error

mdumont-arcom, 2022-06-07:
Hi, 
i follow the steps for build with STM32Cube, when i compile the soft i have this message : 

`../Src/ota_pal/ota_pal_stm32u5_ntz.c:28:10: fatal error: logging_levels.h: No such file or directory`

where is the problem ?

thanks

paulbartell, 2022-06-07:
@mdumont-arcom : Here are a few things to check/ try:

1. Check that your include path includes the "Common/cli" directory as so:
```
"${workspace_loc:/${ProjName}/Common/cli}"
```
You can access this in STM32CubeIDE from the project menu at: **Project**->**Properties**->**C/C++ Build**->**Settings**->**Tool Settings**->**MCU GCC Compiler**->**Include Paths**

2. Verify that your STM32CubeIDE workspace is set to the root directory of this repository. If not, the location references to the libraries and Common directory will need adjustment.

3. Close STM32CubeIDE, then revert the .cproject and .project files located in each project to the version in this repository, and finally re-open STM32CubeIDE.
```
git checkout origin/main Projects/b_u585i_iot02a_tfm/.project
git checkout origin/main Projects/b_u585i_iot02a_tfm/.cproject
git checkout origin/main Projects/b_u585i_iot02a_ntz/.project
git checkout origin/main Projects/b_u585i_iot02a_ntz/.cproject
```

If none of these address your build issue, please provide more lines of your build output log near the error message and list which Operating System and version of STM32CubeIDE you are using so I can attempt to reproduce the problem.

mdumont-arcom, 2022-06-08:
It's good that compiles, I had not understood that it was necessary to fix the workspace at the source of the repository. Once done, the compilation runs smoothly.

closed 2022-06-08 by mdumont-arcom