FreeRTOS/iot-reference-stm32u5 #62: Tickless idle support

nilesh-dryad, 2022-06-07:
Is there any plan to add tickless idle support with STOP mode?
I can see there are two examples of STM32U585. I can see that this repo supports other U5 series also. It would be good if you add some template example for STM32U575 or 95.

paulbartell, 2022-06-07:
@nilesh-dryad : Thanks for your request. I will discuss it with the AWS and ST team. Unfortunately, I cannot make any timeline commitments.

Do you have a particular target board / application in mind?

nilesh-dryad, 2022-06-08:
Thank you for your reply.
I am using Nucleo U575ZI-Q. I have already made one simple example but I did by porting freertos by myself. It is working upto one level. If I only use WFI instruction in IDLE task then it works. But If I use STOP1 mode, then it doesn't work. Looks like FreeRTOS PendSV handler is not able to wake up the STOP1 mode.

When it works, then also it wakes up at every 250ms for no reason. I also don't understand it. I am sending one char on ITM console and then 500ms of busy wait and 5 second of FreeRTOS block. As per below image, it is working but there is wakeup at every 250-300ms which I don't understand. Therefore I ask you to add such example because U5XX series is ultra low power controller. People would expect to have tickless functionality with STOP1 and STOP2 support.

[image]

nilesh-dryad, 2022-06-09:
EDIT: I found that periodic wake up in above image is not actually periodic wakeup,it is the noise from STM32 POWER monitor board. I toggled one GPIO before entering and exiting from Sleep mode in freertos port.c. Look at the below image. When the GPIO is low, it is in sleep mode, and when it is high, it is out of the sleep mode.

> static void vHeartbeatTask( void * pvParameters )
{
    while( 1 )
    {
        vTaskDelay( pdMS_TO_TICKS(5000 ) );
        Task_action('1');
        HAL_Delay(500);
    }
}

[image]

paulbartell, 2022-07-27:
@nilesh-dryad : I'm glad you found the source of your issue. If you would like to share your tickless idle setup, feel free to open a pull request.

closed 2022-07-27 by paulbartell