FreeRTOS/iot-reference-stm32u5 #69: Build issues with STM32CubeIDE

Mimmo-c, 2022-07-10:
Hi,
When I build **b_u585i_iot02a_tfm** with STM32CubeIDE 1.9.0 I got the following errors:
```
Calling TFM build for artifact: tfm_build/bin/bl2.axf due to prereq: tfm_build/.ready
make[3]: *** No rule to make target 'all'.  Stop.
make[3]: *** No rule to make target 'tfm_build/install/interface/lib/s_veneers.o', needed by '../tfm/interface/libtfm_interface.a'.  Stop.
make[2]: *** [../tfm.mk:213: tfm_build/bin/bl2.axf] Error 2
make[1]: *** [../makefile.targets:6: nspe_pre_build] Error 2
make: *** [makefile:100: all] Error 2
"make -j4 all" terminated with exit code 2. Build might be incomplete.
```
I cannot find where is the problem.
Thanks

paulbartell, 2022-07-27:
@Mimmo-c : Apologies for the delayed response. 

The project is built with STM32CubeIDE 1.9 on windows as part of our Github Actions setup. You can review the [build.yml](https://github.com/FreeRTOS/iot-reference-stm32u5/blob/8c02f071f32db51bd845fbcdf8b2e972794fc3b5/.github/workflows/build.yaml#L91) file for some concise setup instructions.

The error you listed would indicate that cmake failed to setup your TF-M build directory. Usually this is due to a missing prerequisite of some kind.

Can you run through the instructions in the [Getting Started Guide](https://github.com/FreeRTOS/iot-reference-stm32u5/blob/main/Getting_Started_Guide.md) and make sure nothing was missed?

If the problem persists, please run a clean build and include more of the logs in a response so I can look into it further.

Mimmo-c, 2022-08-15:
Hi @paulbartell,
Sorry for the delay in responding. I followed all the [Getting Started Guide](https://github.com/FreeRTOS/iot-reference-stm32u5/blob/main/Getting_Started_Guide.md) steps but still have errors in the compilation; now I'm using STM32CubeIDE 1.10.1.
I report below the console log.

`make -j4 all 
make -f ../tfm.mk info spe_bin nspe_build_reqs
PROJECT_NAME:     b_u585i_iot02a_tfm
WORKSPACE_PATH:   C:/lab-iot-reference-stm32u5
BUILD_PATH:       .
TOOLS_PATH:       C:/lab-iot-reference-stm32u5/tools
MIDDLEWARE_PATH:  C:/lab-iot-reference-stm32u5/Middleware
PROJECT_PATH:     ..
TFM_BUILD_PATH:   ./tfm_build
TFM_SRC_PATH:     C:/lab-iot-reference-stm32u5/Middleware/ARM/trusted-firmware-m
MBEDTLS_SRC_PATH: C:/lab-iot-reference-stm32u5/Middleware/ARM/mbedtls
MCUBOOT_SRC_PATH: C:/lab-iot-reference-stm32u5/Middleware/ARM/mcuboot
SHELLFLAGS:       -c
CFLAGS:           
LDFLAGS:          
SHELL:            C:/ST/STM32CubeIDE_1.10.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.make.win32_2.0.100.202202231230/tools/bin/sh.exe
Applying mbedtls patch: 0003-Disable-export-MbedTLSTargets.patch
Applying mbedtls patch: 0002-Enable-crypto-code-sharing-between-independent-binar.patch
Applying mbedtls patch: 0001-BUILD-Update-IAR-support-in-CMakeLists.txt.patch
Calling cmake for artifact: tfm_build/.ready due to prereq: C:/lab-iot-reference-stm32u5/Middleware/ARM/mbedtls/0003-Disable-export-MbedTLSTargets.patched C:/lab-iot-reference-stm32u5/Middleware/ARM/mbedtls/0002-Enable-crypto-code-sharing-between-independent-binar.patched C:/lab-iot-reference-stm32u5/Middleware/ARM/mbedtls/0001-BUILD-Update-IAR-support-in-CMakeLists.txt.patched
OSTYPE: ms/windows
OSTYPE: ms/windows
c:/st/stm32cubeide_1.10.1/stm32cubeide/plugins/com.st.stm32cube.ide.mcu.externaltools.make.win32_2.0.100.202202231230/tools/bin/sh: cmake: not found
Calling TFM build for artifact: tfm_build/bin/bl2.axf due to prereq: tfm_build/.ready
make[3]: *** No rule to make target 'all'.  Stop.
make[3]: *** No rule to make target 'tfm_build/install/interface/lib/s_veneers.o', needed by '../tfm/interface/libtfm_interface.a'.  Stop.
make[2]: *** [../tfm.mk:213: tfm_build/bin/bl2.axf] Error 2
make[1]: *** [../makefile.targets:6: nspe_pre_build] Error 2
make: *** [makefile:100: all] Error 2
"make -j4 all" terminated with exit code 2. Build might be incomplete.`

Thank you

ActoryOu, 2022-10-07:
Hi @Mimmo-c,
Thanks for your feedback!
Please help to paste the content of below files after compiling error:
- lab-iot-reference-stm32u5\Projects\b_u585i_iot02a_tfm\Debug\tfm_build\CMakeFiles\CMakeError.log
- lab-iot-reference-stm32u5\Projects\b_u585i_iot02a_tfm\Debug\tfm_build\CMakeFiles\CMakeOutput.log

I think you got failure on generating .ready file [here](https://github.com/FreeRTOS/iot-reference-stm32u5/blob/main/Projects/b_u585i_iot02a_tfm/tfm.mk#L159). But I need more clues to find the reason.

Thanks!

Mimmo-c, 2022-10-16:
Hi @ActoryOu,
No file is found in the folder you indicated. I currently have STM32CubeIDE version 1.10.1 installed, is it okay if I compile the code with this version?
Thanks

ActoryOu, 2022-10-17:
> Hi @ActoryOu, No file is found in the folder you indicated. I currently have STM32CubeIDE version 1.10.1 installed, is it okay if I compile the code with this version? Thanks

Yes, it should be able to compiled with STM32CubeIDE version 1.10.1. Actually, I'm using this version to compile successfully.

Mimmo-c, 2022-10-23:
Hi @ActoryOu,
I tried again to compile the coide but it crashes and the folders you indicated remain empty.
Thank you

ActoryOu, 2022-10-26:
Hi @Mimmo-c,
Could you help to set `DEBUG=1` at the begining of tools/env_setup.sh and run `source tools/env_setup.sh` on your shell?
I suspect that there is something wrong on python environment.

Thanks!

Mubhadiya-eic, 2024-02-12:
Any resolution of this error ? I am also getting same error with STM32CubeIDE Version: 1.9.0
Build: 12015_20220302_0855 (UTC)

paulbartell, 2024-02-15:
@Mubhadiya-eic Can you run through the Getting Started Guide again and check that none of the prerequisites are missing?

amolchor25, 2024-03-06:
Hello team !!

I have got the same issue, I have followed the [Getting_Started_Guide](https://github.com/FreeRTOS/iot-reference-stm32u5/blob/main/Getting_Started_Guide.md).

below is the log 

```
08:36:47 **** Build of configuration Debug for project b_u585i_iot02a_tfm ****
make -j8 all 
make -f ../tfm.mk info spe_bin nspe_build_reqs
PROJECT_NAME:     b_u585i_iot02a_tfm
WORKSPACE_PATH:   C:/Users/111439/iot-reference-stm32u5
BUILD_PATH:       .
TOOLS_PATH:       C:/Users/111439/iot-reference-stm32u5/tools
MIDDLEWARE_PATH:  C:/Users/111439/iot-reference-stm32u5/Middleware
PROJECT_PATH:     ..
TFM_BUILD_PATH:   ./tfm_build
TFM_SRC_PATH:     C:/Users/111439/iot-reference-stm32u5/Middleware/ARM/trusted-firmware-m
MBEDTLS_SRC_PATH: C:/Users/111439/iot-reference-stm32u5/Middleware/ARM/mbedtls
MCUBOOT_SRC_PATH: C:/Users/111439/iot-reference-stm32u5/Middleware/ARM/mcuboot
SHELLFLAGS:       -c
CFLAGS:           
LDFLAGS:          
SHELL:            C:/ST/STM32CubeIDE_1.12.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.make.win32_2.0.200.202301161003/tools/bin/sh.exe
Applying mbedtls patch: 0003-Disable-export-MbedTLSTargets.patch
Applying mbedtls patch: 0002-Enable-crypto-code-sharing-between-independent-binar.patch
Applying mbedtls patch: 0001-BUILD-Update-IAR-support-in-CMakeLists.txt.patch
Calling cmake for artifact: tfm_build/.ready due to prereq: C:/Users/111439/iot-reference-stm32u5/Middleware/ARM/mbedtls/0003-Disable-export-MbedTLSTargets.patched C:/Users/111439/iot-reference-stm32u5/Middleware/ARM/mbedtls/0002-Enable-crypto-code-sharing-between-independent-binar.patched C:/Users/111439/iot-reference-stm32u5/Middleware/ARM/mbedtls/0001-BUILD-Update-IAR-support-in-CMakeLists.txt.patched
OSTYPE: ms/windows
OSTYPE: ms/windows
-- The C compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/ST/STM32CubeIDE_1.12.1/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.10.3-2021.10.win32_1.0.200.202301161003/tools/bin/arm-none-eabi-gcc.exe
-- Found Python3: C:/Users/111439/iot-reference-stm32u5/.venv/Scripts/python.exe (found version "3.12.2") found components: Interpreter
Start to generate partition files:
Generating Header: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/protected_storage/psa_manifest/tfm_protected_storage.h
Generating Intermedia: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/protected_storage/auto_generated/intermedia_tfm_protected_storage.c
Generating Loadinfo: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/protected_storage/auto_generated/load_info_tfm_protected_storage.c
Generating Header: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/internal_trusted_storage/psa_manifest/tfm_internal_trusted_storage.h
Generating Intermedia: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/internal_trusted_storage/auto_generated/intermedia_tfm_internal_trusted_storage.c
Generating Loadinfo: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/internal_trusted_storage/auto_generated/load_info_tfm_internal_trusted_storage.c
Generating Header: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/crypto/psa_manifest/tfm_crypto.h
Generating Intermedia: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/crypto/auto_generated/intermedia_tfm_crypto.c
Generating Loadinfo: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/crypto/auto_generated/load_info_tfm_crypto.c
Generating Header: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/platform/psa_manifest/tfm_platform.h
Generating Intermedia: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/platform/auto_generated/intermedia_tfm_platform.c
Generating Loadinfo: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/platform/auto_generated/load_info_tfm_platform.c
Generating Header: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/initial_attestation/psa_manifest/tfm_initial_attestation.h
Generating Intermedia: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/initial_attestation/auto_generated/intermedia_tfm_initial_attestation.c
Generating Loadinfo: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/initial_attestation/auto_generated/load_info_tfm_initial_attestation.c
Generating Header: C:/Users/111439/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Debug/tfm_build/generated/secure_fw/partitions/firmware_update/psa_manifest/tfm_firmware_update.h
[... cut ...]
make[3]: *** No rule to make target 'tfm_build/install/interface/lib/s_veneers.o', needed by '../tfm/interface/libtfm_interface.a'.  Stop.
make[2]: *** [../tfm.mk:213: tfm_build/bin/bl2.axf] Error 2
make[1]: *** [../makefile.targets:6: nspe_pre_build] Error 2
make: *** [makefile:107: all] Error 2
"make -j8 all" terminated with exit code 2. Build might be incomplete.

08:37:54 Build Failed. 8 errors, 5 warnings. (took 1m:7s.376ms)
```
Does any one have an idea about the build failed in windows platform?

kstribrnAmzn, 2024-07-08:
@amolchor25 and @Mubhadiya-eic this issue bounced around for a while but has found its way to me. I was able to download the STMCube IDE version 1.9 on my Intel Mac and duplicate your error. Below are my logs...

```
10:55:48 **** Build of configuration Debug for project b_u585i_iot02a_tfm ****
make -j11 all 
make -f ../tfm.mk info spe_bin nspe_build_reqs
PROJECT_NAME:     b_u585i_iot02a_tfm
WORKSPACE_PATH:   /Users/kstribrn/STM32CubeIDE/workspace_1.9.0
BUILD_PATH:       .
TOOLS_PATH:       /Users/kstribrn/STM32CubeIDE/workspace_1.9.0/tools
MIDDLEWARE_PATH:  
PROJECT_PATH:     ..
TFM_BUILD_PATH:   ./tfm_build
TFM_SRC_PATH:     
MBEDTLS_SRC_PATH: 
MCUBOOT_SRC_PATH: 
SHELLFLAGS:       -c
CFLAGS:           
LDFLAGS:          
SHELL:            /bin/bash
Calling cmake for artifact: tfm_build/.ready due to prereq:
bash: line 1: /Users/kstribrn/STM32CubeIDE/workspace_1.9.0/tools/env_setup.sh: No such file or directory
/bin/bash: line 4: /Users/kstribrn/STM32CubeIDE/workspace_1.9.0/tools/env_setup.sh: No such file or directory
Calling TFM build for artifact: tfm_build/bin/bl2.axf due to prereq: tfm_build/.ready
make[3]: *** No rule to make target 'all'.  Stop.
mkdir -p "tfm/"
sh -c "touch ./tfm/.generated"
make[3]: *** No rule to make target 'tfm_build/install/interface/lib/s_veneers.o', needed by '../tfm/interface/libtfm_interface.a'.  Stop.
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [../tfm.mk:213: tfm_build/bin/bl2.axf] Error 2
make[1]: *** [../makefile.targets:6: nspe_pre_build] Error 2
make: *** [makefile:64: all] Error 2
"make -j11 all" terminated with exit code 2. Build might be incomplete.

10:55:51 Build Failed. 6 errors, 0 warnings. (took 2s.258ms)
```

I'll be tinkering around with this today and hope to find a resolution for you.

kstribrnAmzn, 2024-07-08:
Well my previous issue was all of my own doing. At least on my machine the python virtual environment needed to be created beforehand as I had a bunch of incorrectly mapped commands. Probably because of older installs. With the `setup_env.sh` script successfully ran I was able to build the project.

I do agree with @ActoryOu's assement. I'll try to dig into that build command and what is going on under the hood in the next day or two.

lewishanly2024, 2024-07-12:
I am having the same build issues:
[ 51%] Built target tfm_app_rot_partition_ps
make[3]: *** [Makefile:136: all] Error 2
make[3]: *** No rule to make target 'tfm_build/install/interface/lib/s_veneers.o', needed by '../tfm/interface/libtfm_interface.a'.  Stop.
make[2]: *** [../tfm.mk:213: tfm_build/bin/bl2.axf] Error 2
make[1]: *** [../makefile.targets:6: nspe_pre_build] Error 2
make: *** [makefile:107: all] Error 2
"make -j4 all" terminated with exit code 2. Build might be incomplete.

lewishanly2024, 2024-07-12:
lewis@lewis-Lenovo-Z50-70:~/GITHUB/iot-reference-stm32u5$ source tools/env_setup.sh 
WORKSPACE_PATH env variable was not defined. Defaulting to PWD.
TOOLS_PATH:         /home/lewis/GITHUB/iot-reference-stm32u5/tools
VENV_PATH:          /home/lewis/GITHUB/iot-reference-stm32u5/.venv
MCUBOOT_SCRIPTS:    /home/lewis/GITHUB/iot-reference-stm32u5/Middleware/ARM/mcuboot/scripts

OSTYPE: gnu/linux
OSTYPE: gnu/linux
Activating vritual environment.
SITE_PACKAGES_PATH: /home/lewis/GITHUB/iot-reference-stm32u5/.venv/lib/python3.8/site-packages
Requirement already satisfied: pip in ./.venv/lib/python3.8/site-packages (24.1.2)
Requirement already satisfied: pyyaml in ./.venv/lib/python3.8/site-packages (from -r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 1)) (6.0.1)
Requirement already satisfied: botocore in ./.venv/lib/python3.8/site-packages (from -r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 2)) (1.25.13)
Requirement already satisfied: boto3==1.22.0 in ./.venv/lib/python3.8/site-packages (from -r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 3)) (1.22.0)
Requirement already satisfied: requests in ./.venv/lib/python3.8/site-packages (from -r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 4)) (2.32.3)
Requirement already satisfied: pyserial in ./.venv/lib/python3.8/site-packages (from -r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 5)) (3.5)
Requirement already satisfied: cryptography in ./.venv/lib/python3.8/site-packages (from -r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 6)) (42.0.8)
Requirement already satisfied: black in ./.venv/lib/python3.8/site-packages (from -r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 7)) (24.4.2)
Requirement already satisfied: pre-commit in ./.venv/lib/python3.8/site-packages (from -r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 8)) (3.5.0)
Requirement already satisfied: pyasn1 in ./.venv/lib/python3.8/site-packages (from -r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 9)) (0.6.0)
Requirement already satisfied: cbor2 in ./.venv/lib/python3.8/site-packages (from -r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 10)) (5.6.4)
Requirement already satisfied: click in ./.venv/lib/python3.8/site-packages (from -r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 11)) (8.1.7)
Requirement already satisfied: jinja2 in ./.venv/lib/python3.8/site-packages (from -r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 12)) (3.1.4)
Requirement already satisfied: imgtool==1.9.0 in ./.venv/lib/python3.8/site-packages (from -r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 13)) (1.9.0)
Requirement already satisfied: jmespath<2.0.0,>=0.7.1 in ./.venv/lib/python3.8/site-packages (from boto3==1.22.0->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 3)) (1.0.1)
Requirement already satisfied: s3transfer<0.6.0,>=0.5.0 in ./.venv/lib/python3.8/site-packages (from boto3==1.22.0->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 3)) (0.5.2)
Requirement already satisfied: intelhex>=2.2.1 in ./.venv/lib/python3.8/site-packages (from imgtool==1.9.0->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 13)) (2.3.0)
Requirement already satisfied: python-dateutil<3.0.0,>=2.1 in ./.venv/lib/python3.8/site-packages (from botocore->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 2)) (2.9.0.post0)
Requirement already satisfied: urllib3<1.27,>=1.25.4 in ./.venv/lib/python3.8/site-packages (from botocore->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 2)) (1.26.19)
Requirement already satisfied: charset-normalizer<4,>=2 in ./.venv/lib/python3.8/site-packages (from requests->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 4)) (3.3.2)
Requirement already satisfied: idna<4,>=2.5 in ./.venv/lib/python3.8/site-packages (from requests->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 4)) (3.7)
Requirement already satisfied: certifi>=2017.4.17 in ./.venv/lib/python3.8/site-packages (from requests->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 4)) (2024.7.4)
Requirement already satisfied: cffi>=1.12 in ./.venv/lib/python3.8/site-packages (from cryptography->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 6)) (1.16.0)
Requirement already satisfied: mypy-extensions>=0.4.3 in ./.venv/lib/python3.8/site-packages (from black->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 7)) (1.0.0)
Requirement already satisfied: packaging>=22.0 in ./.venv/lib/python3.8/site-packages (from black->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 7)) (24.1)
Requirement already satisfied: pathspec>=0.9.0 in ./.venv/lib/python3.8/site-packages (from black->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 7)) (0.12.1)
Requirement already satisfied: platformdirs>=2 in ./.venv/lib/python3.8/site-packages (from black->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 7)) (4.2.2)
[... cut ...]
Requirement already satisfied: six>=1.5 in ./.venv/lib/python3.8/site-packages (from python-dateutil<3.0.0,>=2.1->botocore->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 2)) (1.16.0)
Requirement already satisfied: distlib<1,>=0.3.7 in ./.venv/lib/python3.8/site-packages (from virtualenv>=20.10.0->pre-commit->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 8)) (0.3.8)
Requirement already satisfied: filelock<4,>=3.12.2 in ./.venv/lib/python3.8/site-packages (from virtualenv>=20.10.0->pre-commit->-r /home/lewis/GITHUB/iot-reference-stm32u5/tools/requirements.txt (line 8)) (3.15.4)

Everything appears to setup correctly.

jasonpcarroll, 2024-12-11:
Hi all,

I'm here to try to continue resolving this issue as it has not been updated in a long time. Is there any updates from any of you.

Best,

Jason Carroll