FreeRTOS/iot-reference-stm32u5 #72: [Feature Request] Add support/sample for HTTP

nik-markovic, 2022-09-26:
It would be nice if coreHTTP was supported by a reference implementation showing an example HTTPS connection being made. We need to make an HTTPS REST call before connecting to MQTT, so having a reference implementation would be very helpful.

Thanks,
Nik

RichardBarry, 2022-09-26:
Although not integrated into this project, you will find a set of coreHTTP demos here: https://freertos.org/mqtt/basic-mqtt-example.html

nik-markovic, 2022-09-27:
HTTPS is more complex because it requires handling of the TLS connection. The part where the transport interface and handled is being set up could be board-specific and is too complex in my opinion for the end user to deal with. Also the user has to come up with re-connect logic, error handling etc., for which there are plenty of examples already in the Amazon FreeRTOS project. Ideally, an HTTPS capable client would be provided to witch the user only needs to present a TLS certificate, host/port/url, HTTP method and optional data and a callback with the response. TLS expertise is required to properly implement the MbedTLS connection, server certificate verification against the stored certificate etc., though some of it is probably available in the MQTT example.

The example that you are referring to is very rudimentary and 30 LOC compared to 300 LOC of scaffolding provided in the Amazon FreeRTOS demo here [](https://github.com/aws/amazon-freertos/blob/747f07402a744ec839ed9950e841142408abd6b0/demos/coreHTTP/http_demo_s3_download.c)

What I am trying to say is that each one of us that needs HTTPS connectivity will have to go through the drill of understanding each one of those examples and components in depth that's probably not really needed to "just do a REST call". It is better is this complexity is dealt with by the sample project or some other reference so all could benefit from it. It is of course up to you to gauge and weigh the impact of this potential contribution, I just wanted to clarify, I guess.

Skptak, 2022-10-06:
Hey @nik-markovic, I wanted to reach out and see if you had seen this resource yet: https://www.freertos.org/http/http-demo-with-tls-mutual-authentication.html 
Since you mentioned using a TLS connection I wanted to make sure you had seen the windows simulator demo we have that is designed to do exactly that.  Additionally, here is the link to the relevant demo on github:
https://github.com/FreeRTOS/FreeRTOS/tree/main/FreeRTOS-Plus/Demo/coreHTTP_Windows_Simulator/HTTP_Mutual_Auth
Lastly, this [link to an echo server README](https://github.com/FreeRTOS/FreeRTOS-Libraries-Integration-Tests/blob/main/tools/echo_server/README.md) provides some basic instructions for how a user can set up a local HTTPS server for testing their connections.
I hope this is helpful, if you need any additional guidance or help please reach out.

n9wxu, 2022-11-09:
Skptak provided additional example resources that seem to satisfy the OP's request.  If more help is needed, please reopen the issue.

closed 2022-11-09 by n9wxu