FreeRTOS/iot-reference-stm32u5 #73: [Feature Request] Add support for NTP

nik-markovic, 2022-09-26:
Time synchronization with coreSNTP does not seem to be integrated for system calls like gettimeofday()/time(). Our project is migrating from Amazon FreeRTOS codebase (which supports SNTP) and needs the current system time, and I suspect that other projects could benefit from this feature too.

Also it doesn't seem that we can check for expiration of certificates when connecting to a server, because time is not available.

Thanks,
Nik

Skptak, 2022-10-10:
Hey @nik-markovic , thanks for submitting this feature request. I've reached out to the team that works on coreSNTP and passed this feature request along to them. 
In the meantime I was wondering if you had seen this https://www.freertos.org/sntp/setting-up-an-ntp-server.html resource. Since it seems that you're trying to use NTP to sync time, and a SNTP client can request the time from both NTP and SNTP servers. 
If you successfully port this feature to FreeRTOS AWS Reference Integrations, your contribution back to this repository would be greatly appreciated.

kstribrnAmzn, 2022-12-08:
I suspect this was reopened due to the new label. I'm going to close this once again.

closed 2022-12-08 by kstribrnAmzn