FreeRTOS/iot-reference-stm32u5 #84: [BUG] HardFault - Exception came from non-secure FW in thread mode

dhwalters423, 2023-02-28:
**Describe the bug**
Using v202212.00
After setting the configuration, and generating key and cert onboard and resetting device, receive HardFault (directly after this step: https://github.com/FreeRTOS/iot-reference-stm32u5/blob/main/Getting_Started_Guide.md#reset-the-target-device)

Device ends up in unrecoverable boot loop.

**Host**
- Host OS: MacOS
- Version: Monterey 12.0.6

**To Reproduce**
- Provision/configure device up to step "Reset the target device"
- Device ends in unrecoverable boot loop 

**Expected behavior**
Connect to AWS IoT Core

**Device Logs**
```
[INF] Starting bootloader
[WRN] This device was provisioned with dummy keys. This device is NOT SECURE
[INF] Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
[INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[INF] Boot source: primary slot
[INF] Swap type: none
[INF] Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x1
[INF] Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
[INF] Boot source: primary slot
[INF] Swap type: none
[INF] Bootloader chainload address offset: 0x3a000
[INF] Jumping to the first image slot
[WRN] This device was provisioned with dummy keys. This device is NOT SECURE
[Sec Thread] Secure image initializing!
TF-M isolation level is: 0x00000002
TF-M FP mode: Software
Booting TFM v1.5.0

<INF>        0 [None      ] HW Init Complete. (app_main.c:177)
<INF>        9 [OTAUpdate ] OTA Agent: Secure Image version 1.5.0, Non-secure Image Version: 1.0.0 (ota_update_task.c:1338)
<INF>        9 [OTAUpdate ] Waiting until MQTT Agent is connected. (ota_update_task.c:1367)
<INF>      120 [MQTTAgent ] Client Certificate: CN=stm32u5_001, SN:0x4991B79C5C0E92B0 (mbedtls_transport.c:335)
<INF>      120 [MQTTAgent ] Issuer: CN=stm32u5_001 (mbedtls_transport.c:336)
<INF>      120 [MQTTAgent ] Valid From: 1970-01-01, Expires: 2069-12-31 (mbedtls_transport.c:337)
> FATAL ERROR: HardFault
Here is some context for the exception:
    EXC_RETURN (LR): 0xFFFFFFBD
    Exception came from non-secure FW in thread mode.
    xPSR:    0x20000003
    MSP:     0x30022BF8
    PSP:     0x3002E6F8
    MSP_NS:  0x200BF3F0
    PSP_NS:  0x2005C5B8
    Exception frame at: 0x2005C5B8
        R0:   0x00000201
        R1:   0x00000000
        R2:   0x00000200
        R3:   0x00000201
        R12:  0x0000000A
        LR:   0x080EBCA9
        PC:   0x080EB0F2
        xPSR: 0x21000000
    CFSR:  0x00000000
    BFSR:  0x00000000
    BFAR:  Not Valid
    MMFSR: 0x00000000
    MMFAR: Not Valid
    UFSR:  0x00000000
    HFSR:  0x40000000
    SFSR:  0x00000048
    SFAR: 0x00000201
[INF] Starting bootloader
[WRN] This device was provisioned with dummy keys. This device is NOT SECURE
...<loop>
```

dhwalters423, 2023-02-28:
The hard fault is caused because I did not import the root certificate.
After running the following to import AmazonRootCA3.pem:

```
pki import cert root_ca_cert
-----BEGIN CERTIFICATE-----
MIIBtjCCAVugAwIBAgITBmyf1XSXNmY/Owua2eiedgPySjAKBggqhkjOPQQDAjA5
MQswCQYDVQQGEwJVUzEPMA0GA1UEChMGQW1hem9uMRkwFwYDVQQDExBBbWF6b24g
Um9vdCBDQSAzMB4XDTE1MDUyNjAwMDAwMFoXDTQwMDUyNjAwMDAwMFowOTELMAkG
A1UEBhMCVVMxDzANBgNVBAoTBkFtYXpvbjEZMBcGA1UEAxMQQW1hem9uIFJvb3Qg
Q0EgMzBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IABCmXp8ZBf8ANm+gBG1bG8lKl
ui2yEujSLtf6ycXYqm0fc4E7O5hrOXwzpcVOho6AF2hiRVd9RFgdszflZwjrZt6j
QjBAMA8GA1UdEwEB/wQFMAMBAf8wDgYDVR0PAQH/BAQDAgGGMB0GA1UdDgQWBBSr
ttvXBp43rDCGB5Fwx5zEGbF4wDAKBggqhkjOPQQDAgNJADBGAiEA4IWSoxe3jfkr
BqWTrBqYaGFy+uGh0PsceGCmQ5nFuMQCIQCcAu/xlJyzlvnrxir4tiz+OpAUFteM
YyRIHN8wfdVoOw==
-----END CERTIFICATE-----
```

The example ran successfully. 
These instructions should be added to the getting started guide. I also am not sure if HardFault is the expected behavior if the server CA cert is missing.

dachalco, 2023-03-01:
@dhwalters423

Thank you for reporting this along with your solution. I suspect a hard fault wasn't the intended behavior here. @paulbartell Can you take a look?

paulbartell, 2023-03-01:
@dhwalters423 Thanks for the report. I am looking into the issue.

dhwalters423, 2023-05-09:
@paulbartell Has there been any update on this issue?

commit d597b870 (2023-07-12): PkiObject: Replace 64-bit type printf specifiers

Replace 64-bit printf specifiers with two 32-bit specifiers for compatibiltiy with newlibc nano printf.

Fixes #84
commit 6fe71ba1 (2023-07-14): PkiObject: Replace 64-bit type printf specifiers

Replace 64-bit printf specifiers with two 32-bit specifiers for compatibiltiy with newlibc nano printf.

Fixes #84
commit 844cf5a6 (2023-07-14): mbedtls_transport: Use result of xConfigureCAChain

Check the result of calling xConfigureCAChain before calling
mbedtls_ssl_conf_ca_chain.

fixes #84
closed 2023-07-14 by paulbartell