FreeRTOS/iot-reference-stm32u5 #94: [BUG] tools/provision.py not compatible with python cryptography module >= 41

grdSTM, 2024-02-20:
**Describe the bug**

Generating device credentials through tools/provision.py and the version of the python cryptography module pulled from tools/requirements.txt fails with the following error message (error returned by load_pem_public_key()):

  Generating a new public/private key pair
  Error: Could not parse public key.

**Host**
- Host OS: Win10 + git-bash
- Python Version: 3.12.2, cryptography 41.x.x or 42.x.x

**To Reproduce**
- Source tools/env_setup.sh in git-bash
- ./tools/provision.py -i (against a programmed device)

**Expected behavior**
- key validation and registration at AWS IoT Core

**Additional context**

Possible workaround:

Holding the cryptography version solves (or masks?) the issue:

diff --git a/tools/requirements.txt b/tools/requirements.txt
index d2f1ade..fb42cca 100644
--- a/tools/requirements.txt
+++ b/tools/requirements.txt
@@ -3,7 +3,7 @@ botocore
 boto3==1.22.0
 requests
 pyserial
-cryptography
+cryptography==40.0.0
 black
 pre-commit
 pyasn1

kstribrnAmzn, 2024-02-22:
Thanks for reporting this @grdSTM! I'm going to try this out on my Mac and see if I can reproduce the same issue. I'd be a bit surprised if this was confined to only windows.

My setup is:
* macOS 13.6.4 (22G513) (Ventura)
* Python 3.11.7
* Python cryptography 41.0.4

kstribrnAmzn, 2024-02-22:
```
Generating a new public/private key pair
Error: Could not parse public key.
```
Able to confirm the bug with my setup. I will begin to work on a fix.

kstribrnAmzn, 2024-02-22:
Thanks to some pointers from @paulbartell and @AniruddhaKanhere we were able to get the problem sorted out. Looks like the public key being returned on [this line](https://github.com/FreeRTOS/iot-reference-stm32u5/blob/main/tools/provision.py#L784) (being generated by the board) in incorrectly formatted and can't be handled by the cryptography library.

An example public key will look something like...
```
-----BEGIN PUBLIC KEY-----
MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEzIY+AnNo27YZn1UuQpP87YaiUy3t
jYi4ieLiqS4p8tO4ejCJQpC8lbUekS8jIfWjU/80dgvXWK2W4OawMO28SgA=
-----END PUBLIC KEY-----
```

When really it should look like...
```
-----BEGIN PUBLIC KEY-----
MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAEzIY+AnNo27YZn1UuQpP87YaiUy3t
jYi4ieLiqS4p8tO4ejCJQpC8lbUekS8jIfWjU/80dgvXWK2W4OawMO28Sg==
-----END PUBLIC KEY-----
```

Notice the slight difference on the last characters of the key. `A=` should be `==`. I'll raise a PR to fix this.

kstribrnAmzn, 2024-02-22:
You can verify this yourself by printing out the returned public key, saving it to a file, and then running `openssl pkey -inform PEM -pubin -in  YOUR_PUB_KEY.pem`

closed 2024-02-23 by kstribrnAmzn

grdSTM, 2024-02-23:
Thanks @kstribrnAmzn, @paulbartell and @AniruddhaKanhere for the root fix!