FreeRTOS/iot-reference-stm32u5 #96: [Feature Request] Add support for lfs on internal nor flash

egoltzman, 2024-04-08:
I have U5 board without external flash and I want to use this package on it but the lfs support semes a must and I was not able to switch to the  lfs_port_internal_nor.c by using pxInitializeInternalFlashFs with out getting hard fault at fs_init.

An alternative will be to use pkcs11 without lfs

xuelix, 2024-04-09:
Thanks for the feature request, I've sent it to the team for consideration.

For a kind reminder, it is preferred to use FreeRTOS forum to post a feature request.  There might be already someone had the solution. 

Here is the link to the forum:  https://forums.freertos.org/

SlimJallouli, 2024-12-13:
Here is the fix for the pxInitializeInternalFlashFs with out getting hard fault at fs_init

https://github.com/FreeRTOS/iot-reference-stm32u5/blob/main/Projects/b_u585i_iot02a_ntz/Src/fs/lfs_port_internal_nor.c#L157