FreeRTOS/iot-reference-stm32u5 #99: Not able to build  CMake Error in C:/BaseIot/iot-reference-stm32u5/Middleware/ARM/mbedtls/library/CMakeLists.txt:   Target "crypto_service_mbedcrypto"

vsdawalkar2511, 2025-05-05:
we stm585 discovery boared and we are clone https://github.com/FreeRTOS/iot-reference-stm32u5/blob/main/
Also we setup enviroment as per Getting_Started_Guide.md
but we are facing build issue for " b_u585i_iot02a_tfm ". Get attached log file
Stm32Ide version 1.17.o
Toolchain 1.13
Would you share any dependency for building code.

Below are errors
-- Configuring done (9.9s)
CMake Error in C:/BaseIot/iot-reference-stm32u5/Middleware/ARM/mbedtls/library/CMakeLists.txt:
  Target "crypto_service_mbedcrypto" INTERFACE_INCLUDE_DIRECTORIES property
  contains path:

    "C:/BaseIot/iot-reference-stm32u5/Middleware/ARM/trusted-firmware-m/secure_fw/partitions/crypto"

  which is prefixed in the source directory.

CMake Error in C:/BaseIot/iot-reference-stm32u5/Middleware/ARM/mbedtls/library/CMakeLists.txt:
  Target "crypto_service_mbedcrypto" INTERFACE_INCLUDE_DIRECTORIES property
  contains path:

    "C:/BaseIot/iot-reference-stm32u5/Middleware/ARM/trusted-firmware-m/platform/ext/accelerator/stm/../../../ext/target/stm/b_u585i_iot02a/accelerator/"

  which is prefixed in the source directory.

CMake Error in C:/BaseIot/iot-reference-stm32u5/Middleware/ARM/mbedtls/library/CMakeLists.txt:
  Target "crypto_service_mbedcrypto" INTERFACE_INCLUDE_DIRECTORIES property
  contains path:

    "C:/BaseIot/iot-reference-stm32u5/Middleware/ARM/trusted-firmware-m/platform/ext/accelerator/stm/../../../ext/target/stm/common/hal/accelerator/"

  which is prefixed in the source directory.

CMake Error in C:/BaseIot/iot-reference-stm32u5/Middleware/ARM/mbedtls/library/CMakeLists.txt:
  Target "crypto_service_mbedcrypto" INTERFACE_INCLUDE_DIRECTORIES property
  contains path:

    "C:/BaseIot/iot-reference-stm32u5/Middleware/ARM/trusted-firmware-m/platform/ext/accelerator/stm/../../../ext/target/stm/b_u585i_iot02a/include/"

  which is prefixed in the source directory.

CMake Error in C:/BaseIot/iot-reference-stm32u5/Middleware/ARM/mbedtls/library/CMakeLists.txt:
  Target "crypto_service_mbedcrypto" INTERFACE_INCLUDE_DIRECTORIES property
  contains path:

    "C:/BaseIot/iot-reference-stm32u5/Middleware/ARM/trusted-firmware-m/platform/ext/accelerator/stm/../../../ext/target/stm/common/stm32u5xx/hal/Inc/"

  which is prefixed in the source directory.

CMake Error in C:/BaseIot/iot-reference-stm32u5/Middleware/ARM/mbedtls/library/CMakeLists.txt:
  Target "crypto_service_mbedcrypto" INTERFACE_INCLUDE_DIRECTORIES property
  contains path:

    "C:/BaseIot/iot-reference-stm32u5/Middleware/ARM/trusted-firmware-m/platform/ext/accelerator/stm/../../../ext/target/stm/common/stm32u5xx/Device/Include/"

  which is prefixed in the source directory.

CMake Error in C:/BaseIot/iot-reference-stm32u5/Middleware/ARM/mbedtls/library/CMakeLists.txt:
  Target "crypto_service_mbedcrypto" INTERFACE_INCLUDE_DIRECTORIES property
  contains path:

    "C:/BaseIot/iot-reference-stm32u5/Middleware/ARM/trusted-firmware-m/platform/ext/accelerator/stm/../../../ext/cmsis"

  which is prefixed in the source directory.

kar-rahul-aws, 2025-05-14:
Hi @vsdawalkar2511 
Thank you for reporting the issue. We are trying to reproduce it. 

In the meanwhile, can you confirm whether all Git submodules have been successfully cloned and initialized?
To be absolutely certain , can you try running 
```
git submodule update --init --recursive 
```
once and trying a clean build. 

Also, since you are using a Windows Environment, Enable the git longpaths option
```
git config --system core.longpaths true
```

tony-josi-aws, 2025-05-22:
@vsdawalkar2511 

Did you get a chance check @kar-rahul-aws's suggestion?

vsdawalkar2511, 2025-05-22:
@tony-josi-aws 
>> Did you get a chance check @kar-rahul-aws's suggestion?
yes but still facing issue. 
Is there is any patch or modification need to build tfm project "b_u585i_iot02a_tfm"

kar-rahul-aws, 2025-05-25:
@vsdawalkar2511 
Is it still the same error that you are facing as you mentioned in the issue description?

vsdawalkar, 2025-05-26:
@kar-rahul-aws 
> [@vsdawalkar2511](https://github.com/vsdawalkar2511) Is it still the same error that you are facing as you mentioned in the issue description?

Yes 
Is there is any patch or modification need to build tfm project "b_u585i_iot02a_tfm" ?

ActoryOu, 2025-06-05:
Hi @vsdawalkar,
Would you mind trying below method by changing tool chain back to 10.3-2021.10?
- https://community.st.com/t5/stm32cubeide-mcus/build-error/m-p/717863/highlight/true#M30491

In short, please replace your toolchain by:
1. open CubeIDE
2. right click on TFM project -> properties -> C/C++ Build -> Settings -> Tool Settings -> MCU/MPU Toolchain
3. Open Toolchain Manager
4. After searching, install 10.3-2021.10
5. IDE asks for a restart after installation
6. Then configure the tool chain from step 2, then selected Fixed -> "GNU Tools for STM32 (10.3-2021.10)"

The root cause is still under investigated. Wanted to know if this helps.

Thank you.

vsdawalkar2511, 2025-06-05:
Hi @ActoryOu 
Yes we try out with 11. 3 tool chain .
But still we facing same issue.
One dout is there is no system file from base code clone from  git "https://github.com/FreeRTOS/iot-reference-stm32u5/blob/main"
 Is there is any precode required or any  patch required ?
if yes then please share

ActoryOu, 2025-06-06:
Hi @vsdawalkar,
Thanks for reporting back.
I do not think it is necessary to provide any extra files to compile this project.

Could you tell us what point did you reach after switching toolchain to 10.3-2021.10 version?

Thank you.

ActoryOu, 2025-06-06:
I'm using MAC to compile this project. And I can get the project built (it shows some error, but I got all binaries I need) after doing below.
1. Follow the steps above to switch toolchain to 10.3-2021.10 version
2. Apply below diff files to the tools
    - tools/requirements
    - tools/env_setup.sh
3. update the cffi to arm64
    - source .venv/bin/activate   (to activate virtual env)
    - pip3 uninstall cffi -y
    - pip3 install --no-binary :all: cffi --ignore-installed
4. build on IDE again

diff of tools/requirements:
```
diff --git a/tools/requirements.txt b/tools/requirements.txt
index d2f1ade..0525093 100644
--- a/tools/requirements.txt
+++ b/tools/requirements.txt
@@ -11,3 +11,4 @@ cbor2
 click
 jinja2
 imgtool==1.9.0
+cmake
```

diff of tools/env_setup.sh:
```
diff --git a/tools/env_setup.sh b/tools/env_setup.sh
index 86e9d36..5d1ba82 100755
--- a/tools/env_setup.sh
+++ b/tools/env_setup.sh
@@ -24,6 +24,7 @@
 #  https://www.FreeRTOS.org
 #  https://github.com/FreeRTOS
 #
+DEBUG=1
 
 [ -n "${WORKSPACE_PATH}" ] || {
     echo "WORKSPACE_PATH env variable was not defined. Defaulting to PWD."
@@ -34,17 +35,17 @@ TOOLS_PATH="${WORKSPACE_PATH}/tools"
 
 [ -n "${DEBUG}" ] && echo "TOOLS_PATH:         ${TOOLS_PATH}"
 
-VENV_PATH=$(realpath "${WORKSPACE_PATH}"/.venv)
+VENV_PATH=$(realpath "${WORKSPACE_PATH}")/.venv
 [ -n "${DEBUG}" ] && echo "VENV_PATH:          ${VENV_PATH}"
 
 MCUBOOT_SCRIPTS=$(realpath "${WORKSPACE_PATH}"/Middleware/ARM/mcuboot/scripts)
 [ -n "${DEBUG}" ] && echo "MCUBOOT_SCRIPTS:    ${MCUBOOT_SCRIPTS}"
 [ -n "${DEBUG}" ] && echo
 
-pycmd="python"
+pycmd="python3"
 
 if command -v python > /dev/null 2>&1; then
-    pycmd="python"
+    pycmd="python3"
 elif command -v python3 > /dev/null 2>&1; then
     pycmd="python3"
 fi
```

Please try steps above and let us know if that works for you.

Thank you.