FreeRTOS/iot-reference-stm32u5 #101: [BUG] Build Errors

virendra-eic, 2025-05-23:
I have cloned iot-reference-stm32u5 repo with and checkout to "SwitchToOTA" branch. If i try to build the project, it does not compiles. There were lots of error of incomplete path, i have resolved them, but could not understand the below error:

/freertos_iot/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Src/ota_pal_tfm.h:31:17: error: unknown type name 'uint16_t'
   31 |                 uint16_t build; /*!< @brief Build of the firmware (Z in firmware version Z.Y.X). */
      |                 ^~~~~~~~
/freertos_iot/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Src/ota_pal_tfm.h:32:17: error: unknown type name 'uint8_t'
   32 |                 uint8_t minor;  /*!< @brief Minor version number of the firmware (Y in firmware version Z.Y.X). */
      |                 ^~~~~~~
/freertos_iot/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Src/ota_pal_tfm.h:34:17: error: unknown type name 'uint8_t'
   34 |                 uint8_t major;  /*!< @brief Major version number of the firmware (X in firmware version Z.Y.X). */
      |                 ^~~~~~~
/freertos_iot/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Src/ota_pal_tfm.h:47:9: error: unknown type name 'uint32_t'
   47 |         uint32_t unsignedVersion32;
      |         ^~~~~~~~
/freertos_iot/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Src/ota_pal_tfm.h:48:9: error: unknown type name 'int32_t'
   48 |         int32_t signedVersion32;
      |         ^~~~~~~
/freertos_iot/iot-reference-stm32u5/Middleware/ARM/ota-pal-psa/ota_pal.c: In function 'otaPal_CheckSignature':
/freertos_iot/iot-reference-stm32u5/Middleware/ARM/ota-pal-psa/ota_pal.c:386:52: warning: pointer targets in passing argument 1 of 'prvConvertToRawECDSASignature' differ in signedness [-Wpointer-sign]
  386 |     if( prvConvertToRawECDSASignature( pFileContext->signature, pFileContext->signatureLen, ucECDSARAWSignature ) == false )
      |                                        ~~~~~~~~~~~~^~~~~~~~~~~
      |                                                    |
      |                                                    const char *
arm-none-eabi-gcc "/freertos_iot/iot-reference-stm32u5/Middleware/ARM/mbedtls/library/aesce.c" -mcpu=cortex-m33 -std=gnu11 -g -DTFM_PSA_API -DBL2 -DBL2_HEADER_SIZE=0x400 -DBL2_TRAILER_SIZE=0x2000 -DDAUTH_CHIP_DEFAULT -DMCUBOOT_IMAGE_NUMBER=2 -DMCUBOOT_SIGN_RSA -DMCUBOOT_SIGN_RSA_LEN=3072 -DTFM_PARTITION_CRYPTO -DTFM_PARTITION_INTERNAL_TRUSTED_STORAGE -DTFM_PARTITION_LOG_LEVEL=TFM_PARTITION_LOG_LEVEL_INFO -DTFM_PARTITION_PLATFORM -DTFM_PARTITION_PROTECTED_STORAGE -DTFM_PSA_API -DTFM_SP_LOG_RAW_ENABLED -DUSE_HAL_DRIVER -DAUTH_CHIP_DEFAULT -DPS_ENCRYPTION -DTFM_PARTITION_PROTECTED_STORAGE -DTFM_PARTITION_INTERNAL_TRUSTED_STORAGE -DTFM_PARTITION_CRYPTO -DTFM_PARTITION_PLATFORM -DTFM_PARTITION_INITIAL_ATTESTATION -DTFM_LVL=2 -DDEBUG -DSTM32 -DB_U585I_IOT02A -DSTM32U585AIIxQ -DSTM32U5 -DSTM32U585xx -DUNITY_INCLUDE_CONFIG_H '-DMBEDTLS_CONFIG_FILE="mbedtls_config_psa.h"' -c -I"/freertos_iot/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/Inc" -I"/freertos_iot/iot-reference-stm32u5/Drivers/STM32U5xx_HAL/Inc" -I"/freertos_iot/iot-reference-stm32u5/Middleware/ARM/mbedtls/library" -I"/freertos_iot/iot-reference-stm32u5/Middleware/ARM/mbedtls/include" -I"/freertos_iot/iot-reference-stm32u5/Projects/b_u585i_iot02a_tfm/tfm/interface/include" -I"/freertos_iot/iot-reference-stm32u5/Drivers/CMSIS/Core/Include" -I"/freertos_iot/iot-reference-stm32u5/Drivers/CMSIS/Device/ST/STM32U5xx/Include" -I"/freertos_iot/iot-reference-stm32u5/Common/cli" -I"/freertos_iot/iot-reference-stm32u5/Middleware/FreeRTOS/kernel/portable/GCC/ARM_CM33_NTZ/non_secure" -I"/freertos_iot/iot-reference-stm32u5/Middleware/FreeRTOS/kernel/include" -I"/freertos_iot/iot-reference-stm32u5/Common/include" -I"/freertos_iot/iot-reference-stm32u5/Common/config" -I"/freertos_iot/iot-reference-stm32u5/Common" -O0 -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Libraries/mbedtls/library/aesce.d" -MT"Libraries/mbedtls/library/aesce.o" --specs=nano.specs -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb -o "Libraries/mbedtls/library/aesce.o"
/freertos_iot/iot-reference-stm32u5/Middleware/ARM/ota-pal-psa/ota_pal.c:116:60: note: expected 'const uint8_t *' {aka 'const unsigned char *'} but argument is of type 'const char *'
  116 | static bool prvConvertToRawECDSASignature( const uint8_t * pucEncodedSignature, size_t signatureLength, uint8_t * pucRawSignature )
      |                                            ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
/freertos_iot/iot-reference-stm32u5/Middleware/ARM/ota-pal-psa/ota_pal.c:392:17: warning: assignment to 'uint8_t *' {aka 'unsigned char *'} from incompatible pointer type 'uint8_t (*)[64]' {aka 'unsigned char (*)[64]'} [-Wincompatible-pointer-types]
  392 |     ucSigBuffer = &ucECDSARAWSignature;
      |                 ^
make[1]: *** [Libraries/ota-pal-psa/subdir.mk:19: Libraries/ota-pal-psa/ota_pal.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [makefile:111: all] Error 2

I am using STM32 Cube IDE version 1.12.1 on linux PC. I have attached the file that contains full logs.

Please help me with this issue.

[build_errors.txt](https://github.com/user-attachments/files/20412110/build_errors.txt)

kar-rahul-aws, 2025-05-25:
Hi @virendra-eic 
Normally the ST library header files and CMSIS pull in stdint.h properly, so these errors should never occur. 
In my Windows environment, the definitions for `uint16_t` and `uint32_t` lie in the path below 
```
C:\ST\STM32CubeIDE_1.16.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.12.3.rel1.win32_1.0.200.202406191623\tools\arm-none-eabi\include\sys\_stdint.h
```

You can try importing some small example for your MCU (using the CubeIDE import example tool) and build it. If the same issue persists , then you can try reinstalling the program.

ActoryOu, 2025-06-07:
Hi @virendra-eic,
Thank you for reporting this issue. The "SwitchToOTA" branch is currently under development and not yet ready for production use. There is some known issues on building this branch, and we're trying to solve it.

Thank you

virendra-eic, 2025-06-09:
Hey @ActoryOu , Thank you for the update!

Thank you!